cnlohr/rv003usb #8: Better instructions parsing of 00/11 to 1/0

Domkeykong, 2023-05-22:
While looking at the rv32e instruction set i found these 2 which can be very useful:
```c
        snez a0, a0 // set not equal zero
```
which can be used here to save 3 instructions:
https://github.com/cnlohr/rv003usb/blob/4cfe8201e15f54e5ece6e178153990f82d764951/firmware/rv003usb.S#L142-L144

and this
```c
        seqz a0, a0 // set equal zero
```
which can be used here as an alternative:
https://github.com/cnlohr/rv003usb/blob/4cfe8201e15f54e5ece6e178153990f82d764951/firmware/rv003usb.S#L285
with both of those changes combined you may even be able to drop the requirements of having D- before D+

cnlohr, 2023-05-23:
Please articulate more, exactly what the change would be (before/after) 

Side-note: I have NEVER seen `sltiu` that's very cool.

cnlohr, 2023-05-23:
side-note: It would be mega cool to be able to do all this on the 24MHz internal clock for other situations w/o pll.

cnlohr, 2023-05-23:
side-side-note:  I really wish they had `c.sltiu`

duk-37, 2023-05-23:
The proposal here is just to replace the highlighted instruction sequences with `snez a0, a0` and `seqz a0, a0` respectively. This works because at either of these points in time `a0` will just be either zero or nonzero. Really the only issue I see with this would be alignment for the rest of the start of `packet_type_loop`, but that's fixable.