cnlohr/rv003usb #10: detachement can be handled without any additional gpio

jnk0le, 2023-05-28:
A trick from f103 usb stacks.

Instead of dedicating another gpio line for attaching by 1,5K resistor, you can tie this resistor to VCC and configure D- line to output logical low for certain amount of time to detach from the bus.

cnlohr, 2023-05-31:
That should certainly be documented and kept, and mentioned in the readme.

xsrf, 2024-01-05:
Added https://github.com/cnlohr/rv003usb/pull/38 which makes `USB_DPU` optional, so you can actually just hardwire D- with 1.5K to 3V3 and force D- low to re-enumerate.

xsrf, 2024-01-27:
Any ideas how long `D-` needs to be held low for re-enumeration? So far couldn't find anything about that in the USB specification. @jnk0le

jnk0le, 2024-01-28:
typically (e.g. stm32duino) it's 10-20ms.

xsrf, 2024-01-28:
> typically (e.g. stm32duino) it's 10-20ms.

I actually found the reference in the USB 2.0 Spec. A disconnect is recognized after only 2.5µs of SE0 (both data lines low). See TDDIS in Table 7-13 of Universal Serial Bus Specification Revision 2.0.

So 10ms is plenty.

cnlohr, 2024-02-01:
Close issue now?  The PR's been merged to make PU resistor is optional.

closed 2024-02-01 by cnlohr