cnlohr/rv003usb #9: Proper connection D-, D+ to CH32V003?

RandoSY, 2023-05-28:
I've been following along, and have successfully built both the ESP32S2 programmer out of an old S2 Mini Di, and also taken a new WCHLinkE, used Zadig to load the proper lib-usb driver, and so my tool chain was able to flash the latest and greatest to a WCHDEV board.

[image]

I was wondering if I can just add a connection to the proper pins on the dev board correctly via D+ and D- from a breakout cable or do I need to make some allowance for voltage levels on the USB interface?

RandoSY, 2023-05-28:
I did find some good info ..

[https://www.beyondlogic.org/usbnutshell/usb2.shtml](url)

[image]

RandoSY, 2023-05-29:
Taking a look at the schematic in Kicad for the development board featured in the Youtube video, on implementing the SW only USB port ... we see places for resistors on the D-, and D+ lines but these are listed as "NP" (not populated).

[image]

cnlohr, 2023-05-29:
DPU is connected, in series with a 1.5k resistor to D- in order to trigger host-detect of the USB device.

mengstr, 2023-06-03:
In the Fritzing "schematic" there are two 15K resistors that pulls both D-lines weakly to the ground. Those are already a part of the host computer and/or hub, so they should not be included in the USB device.

Usually there are  low ohm (68 ohm?) resistors in series with D+ and D- on the USB device. Which I guess have a twofold purpose. They will reduce ringing/reflections on the lines a bit, and as an additional bonus it will prevent bad things happening if both the host and device starts outputting data at the same time. (The latter thing should only happen if the firmware is buggy).

cnlohr, 2023-06-05:
I've used a lot of these, and I typically use ~33 or ~47.

But yeah, they are def a good idea to add!

mengstr, 2023-06-05:
I played around with V-USB for a bit last week on an Arduino mini which I didn't bother to convert to 3.3v. Nor did I add any series resistors or Zeners on the data lines. So straight 5 volts out on the USB bus.

Back in the olden days the USB spec did have a requirement for 5-volt tolerance on the data lines in case of a short in the cabe. But that requirement was dropped in later revisions.

To be on the safe side I connected the device to an old el-cheapo hub which was connected to a USB-A to USB-C hub which I connected to my Mac.  Worked like a charm - I think. I get a 1 ms long "error package" randomly every 500 to 5000ms when it's just sitting there idling. The error package is both lines toggling at 1.5MHz just as something is NRZI encoding a continuous string of zeros.

cnlohr, 2023-10-19:
I've updated the section of the readme here: https://github.com/cnlohr/rv003usb#it-requires-very-little-hardware

closed 2023-10-19 by cnlohr