cnlohr/rv003usb #20: Download Error using minichlink <-> CH32V003 B/D

arondr70, 2023-08-20:
I've been struggling for a week. It's a very basic question, but help me~~ It's blocked here, so it's not moving any further.

*Question : The rv003usb bootloader has been saved to CH32V003 using WCH-LinkE/minichlink. But the below error occurs when the CH32V003(WCH Evolution Board) is then plugged into USB and downloaded using minichlink. What's the problem?

../../ch32v003fun/../minichlink/minichlink -w blink.bin flash -b
Found B003Fun Bootloader
Halting Boot Countdown
Error: Timed out waiting for stub to complete
Could not setup interface.

*Note: Downloading using busbflash in testblink example is normal.

HD: 0x55c1acc1a2a0
Read 40 bytes
Send feature report: 128
Read: 129
Readback Confirm

[Test environment]
* * OS : Ubuntu 20.04 LTS
* * Git Source : 2023/08/20 download (rv003usb/ch32v003fun)

cnlohr, 2023-08-21:
Wow... I don't know what the problem is.  I wonder if there's some pending update to minichlink that was not pushed. I can investigate more tonight.

arondr70, 2023-08-22:
Thank you very much for your reply. If you upload a prebuilt image (bootloader, minichlink) to git to validate the error, I think it will help a lot for a beginner who approaches for the first time like me. Thank you in advance^^

arondr70, 2023-08-27:
* It could be a hardware problem, so I bought an additional CH32V003 board. For this reason, it took time..

* Same as before, I connected as below. (I'm sharing it because I might have made a mistake)
D+ <--33ohm--> PD3
D- <--33ohm--> PD4
PD4 <--1.5K--> PD5

* I downloaded a new source from git because there might be a change.
git clone https://github.com/cnlohr/rv003usb --recursive
cd rv003usb
git submodule update --recurisve

* Running make in the bootloader folder, the build image and the bootloader image with WCH-LinkE were burned successfully. (The log is as follows)
../ch32v003fun/ch32v003fun/../minichlink/minichlink -a -U -w bootloader.bin bootloader -B
Found WCH Link
WCH Programmer is CH32V307 version 2.7
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : fe-2c-ab-cd-66-4a-bc-51
PFlags       : ff-ff-ff-ff
Part Type (B): 00-00-07-24
Read protection: disabled
Interface Setup
FLASH_OBTKEYR = 00000000 (0)
Image written.

* After connecting usb to ch32v003 board, dmesg contents are as follows.
[ 2804.439930] usb 3-6: new low-speed USB device number 59 using xhci_hcd
[ 2804.589834] usb 3-6: config 1 interface 0 altsetting 0 endpoint 0x81 has invalid maxpacket 64, setting to 8
[ 2804.591582] usb 3-6: New USB device found, idVendor=1209, idProduct=b003, bcdDevice= 0.02
[ 2804.591591] usb 3-6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2804.591595] usb 3-6: Product: rv003usb
[ 2804.591599] usb 3-6: Manufacturer: cnlohr
[ 2804.591602] usb 3-6: SerialNumber: NBTT
[ 2804.593856] hid-generic 0003:1209:B003.001C: hiddev1,hidraw4: USB HID v1.10 Device [cnlohr rv003usb] on usb-0000:00:14.0-6/input0

* If you go to the blink example folder and run make, the error will occur the same as the previous query.
../../minichlink/minichlink -w blink.bin flash -b
Found B003Fun Bootloader
Halting Boot Countdown
Error: Timed out waiting for stub to complete
Could not setup interface.
make: *** [../../ch32v003fun/ch32v003fun.mk:66: cv_flash] 오류 223

* I know you're busy, but I'd appreciate it if you could review the progress step by step and let me know what I made a mistake. I want to make something with my daughter, and I really want to use this function.

arondr70, 2023-09-06:
I tested it using two CH32V003 boards, two WCH Links, two computers, and many USB cables, but all of them get the same error. Therefore, it is not a hardware issue.

../../minichlink/minichlink -w blink.bin flash -b
Found B003Fun Bootloader
Halting Boot Countdown
Error: Timed out waiting for stub to complete
Could not setup interface.
make: *** [../../ch32v003fun/ch32v003fun.mk:66: cv_flash] 오류 223

arondr70, 2023-09-06:
After looking at the pgm-b003fun.c source file in the minichlink folder, it is an error message caused by the timeout of the "CommitOp" function during the setup process of the function below. As a result of solving and executing a well-made debugging comment (#define DEBUG_B003), it was confirmed that both the data sent and received were normal.

The normal transmission and reception is also understood as the normal operation of the existing busbflash.c and testtop.c examples.

static int B003FunSetupInterface( void * dev )
{
	struct B003FunProgrammerStruct * eps = (struct B003FunProgrammerStruct*) dev;
	printf( "Halting Boot Countdown\n" );
	ResetOp( eps );
	WriteOpArb( eps, halt_wait_blob, sizeof(halt_wait_blob) );
	if( CommitOp( eps ) ) return -5;   // <----------- timeout error point
	return 0;
}

../../minichlink/minichlink -w blink.bin flash -b
Found B003Fun Bootloader
Halting Boot Countdown
Commit TX: 128 bytes
aa 00 00 00 81 46 94 c1 fd 56 14 c1 82 80 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 cd ab 34 12 

hid_send_feature_report = 128
Commit RX: 128 bytes
aa 00 00 00 81 46 94 c1 fd 56 14 c1 82 80 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 cd ab 34 12

arondr70, 2023-09-06:
It is normal only when the if statement in the "CommitOp" function is satisfied and broken, but it seems that the error occurs because it is timed out.

If the assembly command in the "halt_wait_blob" array is executed, the if statement should be satisfied, but after debugging by adding the GPIO code to the bootloader, the if statement below has not entered. I don't understand the whole thing, but the e->count++ above the if statement looks weird. I only ++ once outside for loop, is this normal? If you enter here, the runwordpad changes in the main function, so the code "halt_wait_blob" will be executed, but this seems to be a problem.

// minichlink/pgm-b003fun.c source
static int CommitOp( struct B003FunProgrammerStruct * eps )
{
.....
		if( eps->respbuffer[1] == 0xff ) break;
....
}

//   li a3, 0; sw a3, 0(a1); li a3, -1; sw a3, 0(a0); ret;
static const unsigned char halt_wait_blob[] = {
	0x81, 0x46, 0x94, 0xc1, 0xfd, 0x56, 0x14, 0xc1, 0x82, 0x80 };

// bootloader.c source
			for( i = 0; i < l; i++ ) 
				start[i] = data[i];//((intptr_t)data)>>(i*8);
			e->count ++;   // <---- I doubt it's an error

			if( e->count >= SCRATCHPAD_SIZE )
			{                     // <----- Failed to enter

arondr70, 2023-09-13:
I solved it. I modified one line of the source code below to confirm that it works normally with minichlink, downloaded to flash and run.

[bootloader.c -- line 264]
if( e->count >= SCRATCHPAD_SIZE )
==> if( e->count >= (SCRATCHPAD_SIZE>>3) )

closed 2023-09-13 by arondr70