cnlohr/rv003usb #23: D+/D- on CH32V003FUN Board?

xsrf, 2023-09-03:
Hey,
so since I have one of your CH32V003FUN boards, I wanted to try the USB there. It's missing the resistors connecting the USB, so I beeped them out:

[image]

I noticed that PD4 is connected to D- and PD3 to D+. But in all your code it's the other way round. Didn't you use the same board during your streams? 🤔

I check the silkscreen on the board, PD3-PD5 are labeled correctly. Am I missing something?

cnlohr, 2023-09-13:
D+/D- are flipped on USB low speed.  In that they mean logically opposite things.  I should have changed the labels :-/

xsrf, 2023-09-13:
> D+/D- are flipped on USB low speed. In that they mean logically opposite things. I should have changed the labels :-/

Well different lines need to be pulled up high, but D+ always is D+ and naming the variable of the pin that is connected to D+ "DN" is kinda confusing.

I got my 0402 resistors yesterday but I'm not at home at the moment. I'll try the board with proper USB on the weekend.

xsrf, 2023-10-21:
@cnlohr now with the schematic and the note about your frustration (which I kind of get if you think in the swapped way) it is still confusing. LS does not swap data-lines, LS and FS just have different pull-ups und thus different idle states.

For Low-Speed, D- needs a 1K5 pull-up. Thus the idle-state / J state is D- HIGH, D+ LOW. 
For Full-Speed, D+ needs a 1K5 pull-up. Thus the idle-state / J state is D- LOW, D+ HIGH.
https://en.wikipedia.org/wiki/USB_communications#Line_transition_state

I get that this looks like D+/D- are swapped. But thinking this way is what makes it confusing.
Also, why do you think of LS being the swapped one? Just think of FS being the swapped one 😎

cnlohr, 2023-12-24:
I'm sorry I checked out for so long.  But it looks like you got this all straightened out.

And, that is fair, but no matter how you think about it being swapped or not it's still confusing >.<

closed 2023-12-24 by cnlohr

xsrf, 2024-02-25:
Resolved with #46