cnlohr/rv003usb #25: 8 pin and 16 pin error

LenarOsx, 2023-09-16:
I worked! its amazing, all examples worked but cdc, com port doesn't work.

I tried the code on ch32v003 over 8pin and 16pin version, with no luck. 
Is there a way to make it work on those versions, or is the code for the 20 pin version only?
Thanks.!

cnlohr, 2023-09-18:
CDC doesn't work on Windows.

There were some recent changes where you can reconfigure it to use other pins.  It should work fine on other variants of the chip.

LenarOsx, 2023-09-25:
I have downloaded the latest code, but it didn't work on 8 and 16 pin version, I found a specific configuration than kind of worked, port A, dm 2, dp 1. It does not show any error like 'invalid usb descriptor', it shows in usb device manager, but it does nothing.
no midi, no hid just shows correctly in the listed devices.

cnlohr, 2023-09-26:
Are those ports reconfigurable? Also did you try swapping them?

mcu23, 2023-10-25:
PC1, PC2, PC4  gamepad demo, Test OK.   

  usb_config.h

```
#define USB_DM 2
#define USB_DP 4
#define USB_DPU 1
#define USB_PORT C
```

[image]

cnlohr, 2023-10-26:
I have linked the above message in the README.  Closing issue now.  Thank you for testing.

dienvx1997bn, 2023-12-26:
> PC1, PC2, PC4 gamepad demo, Test OK.
> 
> usb_config.h
> 
> ```
> #define USB_DM 4
> #define USB_DP 2
> #define USB_DPU 1
>  
> #define USB_PORT C
> ```

i design my own device but not work
#define USB_DM 2
#define USB_DP 4
#define USB_DPU 1
#define USB_PORT C

cnlohr, 2023-12-27:
> > PC1, PC2, PC4 gamepad demo, Test OK.
> > usb_config.h
> > ```
> > #define USB_DM 4
> > #define USB_DP 2
> > #define USB_DPU 1
> >  
> > #define USB_PORT C
> > ```
> 
> i design my own device but not work #define USB_DM 2 #define USB_DP 4 #define USB_DPU 1 #define USB_PORT C

Please clarify - what errors are you receiving on the OS end?  Have you tested this with a known-good board configuration?  Please also share your schematic.  (And it probably makes sense to open a new ticket)

dienvx1997bn, 2023-12-27:
> Please clarify - what errors are you receiving on the OS end? Have you tested this with a known-good board configuration? Please also share your schematic. (And it probably makes sense to open a new [...]

Base on your suggestion design, and use ch32v003 sop8 chip
[image]

cnlohr, 2023-12-27:
I have passed this issue off to some folks in my Discord.

cnlohr, 2023-12-27:
I have passed this issue off to some folks in my Discord, a few who got the SOP-8 version working.  You may want to join that discord and see if they can help more.  I've never used the SOP-8 version of the 003. https://discord.gg/CCeyWyZ

I don't see any reason port C "shouldn't" work, but also, I don't have experience making it work.  Hopefully someone on the discord can shine some light.

closed 2024-02-02 by cnlohr