cnlohr/rv003usb #28: Possible to use PD5/PD6?

machdyne, 2023-10-22:
Hi. I made a custom board that uses PD5/PD6 for DM/DP (oops?) and set the config:

```
#define USB_DM 5
#define USB_DP 6
#define USB_DPU 4
#define USB_PORT D
```

This doesn't seem to work:

../rv003usb/rv003usb.S:56: Error: illegal operands `c.andi a0,((1<<(5))|1<<(6))'

I tried changing all of those instances to ```li t1, USB_DMASK; and a0, a0, t1``` but it apparently doesn't fit in flash.

Any ideas?

mcu23, 2023-10-25:
PC1, PC2, PC4  OK.

Use  this settings at
 usb_config.h

#define USB_DM 2
#define USB_DP 4
#define USB_DPU 1
#define USB_PORT C

[image]

cnlohr, 2023-10-26:
I didn't realize until now that andi can't use pin numbers >= 6. 

It should be possible to make this work, but it will be tricky.  and I don't have availability any time soon.

I will keep that in mind for the future.  I will update the readme.  I'm very sorry for the pain this may cause.

machdyne, 2023-10-26:
Thanks, I'll update this issue if I come up with a solution in the meantime.

cnlohr, 2023-12-18:
Thank you.  I am going to close this issue for now.

closed 2023-12-18 by cnlohr

xsrf, 2024-02-25:
> I didn't realize until now that andi can't use pin numbers >= 6.

Just for the record if someone comes across this, it's actually >= 5.

Only Pins 0-4 can be used for D+/D-

wohltat, 2025-04-18:
It's sad that it doesn't work with PD5/PD6 since there are a lot of cheap boards from aliexpress that have those pin attached to D+ and D-.

Is there another way around that limitation? I have no experience in RISC-V assembly.
Are those parts with `c.andi` very time critical or can it be replaced with something that is a cycle longer?

I've asked ChatGPT and it came up with that solution:

replace
``` asm
c.andi a0, USB_DMASK
```
 with:
``` asm
li t6, USB_DMASK       // use a spare temp reg like t6
andi a0, a0, t6
```

Please let me know if you immediately can see that this cannot work, so that i don't need try any further.

machdyne, 2025-04-18:
The problem I had with that approach was that it no longer fit in flash with all of the added instructions.

wohltat, 2025-04-18:
The board i was talking about is one of those.
[image]

I just cut some traces and did some rewiring and noticed that it would not be enough if PD5 and PD6 can be used.
I missed that there is no voltage regulator to 3.3V, so the ch32 runs on and send 5V directly which makes it fail.
When powering the board externally by 3.3V the USB device shows up.

Seems to me that the guys designing the boards did't understand USB at all since also the D+ and D- line are connected to TX and RX.
Fixing this manually is not worth it, i don't follow that path in favor of a new board.

egormanga, 2025-05-25:
@wohltat Apparently, this was indeed intended to have UART inside the USB-C connector instead of actual USB. It's a pretty common case for lower-cost devices, albeit with different connectors (such as 3.5mm jack) usually.