cnlohr/rv003usb #30: Skip USB_DPU on SO-8?

mikaelsundin, 2023-10-27:
For a long time ago i was experimenting with USB on PIC18F2550 (still got my NES fourscore to USB Gamepad) .

If i remember correctly it is possible to skip the extra GPIO for USB-pullup and connect the 1.5k to 3v3 instead?
To cause a USB re-enumerate the D- pin can be hold low instead by set the GPIO as output low for a little while.

Great job cnlohr btw, you got many interesting repos.

Nable80, 2023-10-28:
Hi!
> If i remember correctly it is possible to skip the extra GPIO for USB-pullup and connect the 1.5k to 3v3 instead?

Yes, it's a viable configuration, it's even documented in README:
> it may connect D- to DPU or VCC if the intent is for the part to always be on-bus

I'm not sure about your second point:
> To cause a USB re-enumerate the D- pin can be hold low instead by set the GPIO as output low for a little while.

I don't think this way is safe, if a host tries to pull this line up during you're holding it down, it'll cause a bus conflict and either host or MCU will be damaged somehow. IMHO (at least) there should be series resistors (10~68R) between the MCU and USB connector, they'll also help with unneeded signal reflections (even low-speed USB may have them).

mikaelsundin, 2023-10-29:
The USB host must be able to survive D+, D- short to GND according to specification. 
So it should be safe to do if the USB host follow the USB specification. 
Short to 5V is removed in "5V Short Circuit Withstand ECN.pdf"

https://www.usb.org/document-library/usb-20-specification
File usb_20.pdf chapter 7.1.1 (page 124)

But a serie resistor is probably wise to add anyway.

I have missed that line in the README. 
Thanks Nable80.

Nable80, 2023-10-29:
Thank you for a reference to the specification!

cnlohr, 2023-10-30:
I would be curious to know more people's feelings about this.  No need to close.

cnlohr, 2023-12-18:
I am going to close this now, but also add the note about a series resistor the README.

closed 2023-12-18 by cnlohr