cnlohr/rv003usb #32: demo_hidapi.c is missing

MartinEssink, 2023-11-05:
The HID API demo cannot be compiled, as it is missing the main c file.

emeb, 2023-12-12:
Any update on this? I'd like to use this demo for some projects.

MartinEssink, 2023-12-12:
I had a look in the commit history, but it seems like the example was added without the file.

emeb, 2023-12-13:
Someone on the cnlohr ch32v003fun discord channel pointed out that the missing file is pretty similar to this one:

https://github.com/cnlohr/rv003usb/blob/6f3ffc4f332ec8057e3c50a529aa708fb914a37f/demo_custom_device/demo_custom_device.c

I copied that into the demo_hidapi directory, changed the name to demo_hidapi.c and stripped out the WS28xx driver code and it appears to work with the host test applications in the testtop dir.

cnlohr, 2023-12-18:
I am terribly sorry, I added the correct one now.

cnlohr, 2023-12-18:
Very sorry again.  But it is resolved, just tested on Windows.  It works. 

I get about 57 KB/s (PC->CH32V003) and 64 KB/s (CH32V003->PC)

closed 2023-12-18 by cnlohr