cnlohr/rv003usb #41: USB Mass Storage support

jonnor, 2024-01-22:
The current code has support/examples for HID class devices. Would it be possible/feasible to implement Mass Storage Class on top of the existing core USB code? Or are there any particular challenges or limitation forseen with that?

My use case is a data logger. I would write to a small (1 MB) FAT32 filesystem to external SPI flash. And then to expose that over USB, to allow copying the logs over to a computer.

cnlohr, 2024-02-01:
It is absolutely theoretically possible, but, last time I tried, neither Windows nor Linux (for different reasons) would let me make a mass storage device, since they really wanted bulk endpoints for the data transfer.

It's totally possible some other descriptor voodoo would work!  I only spent about 15 minutes trying.

cnlohr, 2024-02-01:
Apparently I can't build function rv003usb's here :'( I have some GCC version that doesn't work right.  I guess I'll have to debug tonight.

BogdanTheGeek, 2024-03-06:
Maybe this would help: https://github.com/brendena/pico_drag_n_drop_programmer/blob/main/msc_disk.c

cnlohr, 2024-03-07:
Super curious if that can be made to work with USB-low-speed.