cnlohr/rv003usb #45: How to erase the Bootloader

Saschko2000, 2024-02-23:
Can anyone perhaps explain, how to erase the bootloader from flash?
I want to get rid of the 5sec. wait time at startup.

xsrf, 2024-02-24:
@Saschko2000 good question, I actually came across this question also but so far didn't check if just writing zeroes would be enough or not.

In the meantime, you can just disable it or actually use it with a configuration you like.
I documented the configuration here, which I will merge shortly:
https://github.com/xsrf/rv003usb/tree/Swap_DP_DM/bootloader#bootloader

Saschko2000, 2024-02-24:
Just write zeroes does not work out.
#define BOOTLOADER_TIMEOUT_PWR 1 is a workaround for me.
But I'm still interested to know how to erase it.

xsrf, 2024-02-24:
@Saschko2000 I tried to just read the bootloader partition from a chip I never touched... but it fails. You wanna Try it?
~~~
> minichlink -r boot.bin bootloader 1920

Found WCH Link
WCH Programmer is LinkE version 2.10
Chip Type: 003
Setup success
Flash Storage: 16 kB
Part UUID    : 85-25-ab-cd-ed-49-bc-57
PFlags       : ff-ff-ff-ff
Part Type (B): 07-13-bb-91
Read protection: disabled
Interface Setup
Fault writing memory (DMABSTRACTS = 08000302) (Execption executing Abstract Command) DMSTATUS: 004c0382
Fault on DefaultReadBinaryBlob
Fault reading device
~~~

xsrf, 2024-02-24:
Interesting. Reading 1919 Bytes works, 1920 fails. Here is the factory bootloader:
[ch32v003-factory-bootloader.zip](https://github.com/cnlohr/rv003usb/files/14394135/ch32v003-factory-bootloader.zip)

Saschko2000, 2024-02-24:
Hmm, good idea. Why didn't I think of that?
I tried it, with the same results as you. 1919 works, and 1920 fails. My factory bootloader was identical to yours.
I flashed it and it works fine. 
It is not a erase, but a replace. 
But the main thing is that it works properly.

Thank you very much :-)

closed 2024-02-24 by Saschko2000