cnlohr/rv003usb #49: please add bin file or steps to compiles

anwfeb, 2024-02-26:
how to create bin file for bootloader? please make the steps, or you can add the compiled bin bootloader file. I use CH32V003 by adding hardware components as shown in the instructions. https://github.com/cnlohr/rv003usb?tab=readme-ov-file#it-requires-very-little-hardware

xsrf, 2024-02-26:
You open the bootloader directory in your console and you execute `make`. On Windows you can execute `make_win.bat`. Whenever you change a header file, execute `make clean`.

This should compile and upload the bootloader to a chip via minichlink.

You get any errors?

anwfeb, 2024-02-26:
please check this, I can't do that.

Microsoft Windows [Version 10.0.19045.4046]
(c) Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>cd C:\rv003usb\bootloader

C:\rv003usb\bootloader>make_win.bat

C:\rv003usb\bootloader>REM Bad USB Operation on pgm-wch-linke.c ? Zadig!

C:\rv003usb\bootloader>del bootloader.bin
Could Not Find C:\rv003usb\bootloader\bootloader.bin

C:\rv003usb\bootloader>del bootloader.elf
Could Not Find C:\rv003usb\bootloader\bootloader.elf

C:\rv003usb\bootloader>del bootloader.hex
Could Not Find C:\rv003usb\bootloader\bootloader.hex

C:\rv003usb\bootloader>del bootloader.lst
Could Not Find C:\rv003usb\bootloader\bootloader.lst

C:\rv003usb\bootloader>del bootloader.map
Could Not Find C:\rv003usb\bootloader\bootloader.map

C:\rv003usb\bootloader>make build
riscv64-unknown-elf-gcc -o bootloader.elf  bootloader.c ../rv003usb/rv003usb.S  -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections
process_begin: CreateProcess(NULL, riscv64-unknown-elf-gcc -o bootloader.elf bootloader.c ../rv003usb/rv003usb.S -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [bootloader.elf] Error 2

C:\rv003usb\bootloader>..\ch32v003fun\minichlink\minichlink.exe -a -w bootloader.bin bootloader -B
Error: couldn't open wch link device (libusb_open() = -5)
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers

C:\rv003usb\bootloader>

xsrf, 2024-02-26:
Did you follow the installation of GCC10 from https://github.com/cnlohr/ch32v003fun/wiki/Installation#windows ?
I guess make can't execute `riscv64-unknown-elf-gcc`.
Can you just execute `riscv64-unknown-elf-gcc` from the teminal?
If it's not found and you installed GCC10, make sure your %path% environmen variable contains the GCC10 folder.

anwfeb, 2024-02-27:
i have installed GCC10, but I don't understand what the %path% step you mean is. now I see an error like this.

`Microsoft Windows [Version 10.0.19045.4046]
(c) Microsoft Corporation. All rights reserved.

C:\WINDOWS\system32>cd C:\rv003usb\bootloader

C:\rv003usb\bootloader>make_win.bat

C:\rv003usb\bootloader>REM Bad USB Operation on pgm-wch-linke.c ? Zadig!

C:\rv003usb\bootloader>del bootloader.bin
Could Not Find C:\rv003usb\bootloader\bootloader.bin

C:\rv003usb\bootloader>del bootloader.elf
Could Not Find C:\rv003usb\bootloader\bootloader.elf

C:\rv003usb\bootloader>del bootloader.hex
Could Not Find C:\rv003usb\bootloader\bootloader.hex

C:\rv003usb\bootloader>del bootloader.lst
Could Not Find C:\rv003usb\bootloader\bootloader.lst

C:\rv003usb\bootloader>del bootloader.map
Could Not Find C:\rv003usb\bootloader\bootloader.map

C:\rv003usb\bootloader>make build
riscv64-unknown-elf-gcc -o bootloader.elf  bootloader.c ../rv003usb/rv003usb.S  -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections
../ch32v003fun/ch32v003fun/ch32v003fun.h: Assembler messages:
../ch32v003fun/ch32v003fun/ch32v003fun.h:636: Error: unrecognized opcode `typedef struct'
../ch32v003fun/ch32v003fun/ch32v003fun.h:637: Error: junk at end of line, first unrecognized character is `{'
../ch32v003fun/ch32v003fun/ch32v003fun.h:638: Error: unrecognized opcode `__io uint32_t CFGR0'
../ch32v003fun/ch32v003fun/ch32v003fun.h:639: Error: unrecognized opcode `__io uint32_t CFGR1'
../ch32v003fun/ch32v003fun/ch32v003fun.h:640: Error: junk at end of line, first unrecognized character is `}'
../ch32v003fun/ch32v003fun/ch32v003fun.h:643: Error: unrecognized opcode `typedef struct'
../ch32v003fun/ch32v003fun/ch32v003fun.h:644: Error: junk at end of line, first unrecognized character is `{'
../ch32v003fun/ch32v003fun/ch32v003fun.h:645: Error: unrecognized opcode `__io uint32_t CFGR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:646: Error: unrecognized opcode `__io uint32_t CNTR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:647: Error: unrecognized opcode `__io uint32_t PADDR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:648: Error: unrecognized opcode `__io uint32_t MADDR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:649: Error: junk at end of line, first unrecognized character is `}'
../ch32v003fun/ch32v003fun/ch32v003fun.h:651: Error: unrecognized opcode `typedef struct'
../ch32v003fun/ch32v003fun/ch32v003fun.h:652: Error: junk at end of line, first unrecognized character is `{'
../ch32v003fun/ch32v003fun/ch32v003fun.h:653: Error: unrecognized opcode `__io uint32_t INTFR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:654: Error: unrecognized opcode `__io uint32_t INTFCR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:655: Error: junk at end of line, first unrecognized character is `}'
../ch32v003fun/ch32v003fun/ch32v003fun.h:658: Error: unrecognized opcode `typedef struct'
../ch32v003fun/ch32v003fun/ch32v003fun.h:659: Error: junk at end of line, first unrecognized character is `{'
../ch32v003fun/ch32v003fun/ch32v003fun.h:660: Error: unrecognized opcode `__io uint32_t INTENR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:661: Error: unrecognized opcode `__io uint32_t EVENR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:662: Error: unrecognized opcode `__io uint32_t RTENR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:663: Error: unrecognized opcode `__io uint32_t FTENR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:664: Error: unrecognized opcode `__io uint32_t SWIEVR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:665: Error: unrecognized opcode `__io uint32_t INTFR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:666: Error: junk at end of line, first unrecognized character is `}'
../ch32v003fun/ch32v003fun/ch32v003fun.h:669: Error: unrecognized opcode `typedef struct'
../ch32v003fun/ch32v003fun/ch32v003fun.h:670: Error: junk at end of line, first unrecognized character is `{'
../ch32v003fun/ch32v003fun/ch32v003fun.h:671: Error: unrecognized opcode `__io uint32_t ACTLR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:672: Error: unrecognized opcode `__io uint32_t KEYR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:673: Error: unrecognized opcode `__io uint32_t OBKEYR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:674: Error: unrecognized opcode `__io uint32_t STATR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:675: Error: unrecognized opcode `__io uint32_t CTLR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:676: Error: unrecognized opcode `__io uint32_t ADDR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:677: Error: unrecognized opcode `__io uint32_t RESERVED'
../ch32v003fun/ch32v003fun/ch32v003fun.h:678: Error: unrecognized opcode `__io uint32_t OBR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:679: Error: unrecognized opcode `__io uint32_t WPR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:680: Error: unrecognized opcode `__io uint32_t MODEKEYR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:682: Error: unrecognized opcode `__io uint32_t BOOT_MODEKEYR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:684: Error: junk at end of line, first unrecognized character is `}'
../ch32v003fun/ch32v003fun/ch32v003fun.h:687: Error: unrecognized opcode `typedef struct'
[... 261 lines cut ...]
[... cut ...]
../ch32v003fun/ch32v003fun/ch32v003fun.h:1124: Error: junk at end of line, first unrecognized character is `{'
../ch32v003fun/ch32v003fun/ch32v003fun.h:1125: Error: unrecognized opcode `__io uint32_t EXTEN_CTR'
../ch32v003fun/ch32v003fun/ch32v003fun.h:1126: Error: junk at end of line, first unrecognized character is `}'
make: *** [bootloader.elf] Error 1

C:\rv003usb\bootloader>..\ch32v003fun\minichlink\minichlink.exe -a -w bootloader.bin bootloader -B
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers

C:\rv003usb\bootloader>`

cnlohr, 2024-02-27:
@anwfeb  I am not sure how you got an unpinned ch32v003fun version, but I just fixed the upstream ch32v003fun.

@xsrf I am not in a position to  test the USB stuff for the next day or so can you sync to ch32v003fun master and see if everything's still kosher?

anwfeb, 2024-02-27:
> @anwfeb Saya tidak yakin bagaimana Anda mendapatkan versi ch32v003fun yang tidak dipasangi pin, tetapi saya baru saja memperbaiki ch32v003fun upstream.

i used ch32v003fun from your source code https://github.com/cnlohr/ch32v003fun/tree/f1820f2995e146ce8be0e3a94cb34a53d8238061 but maybe there are some steps missed, i don't understand. i'm just trying to follow the instructions here.

xsrf, 2024-02-27:
So, Windows Build works for me
```
C:\test>git clone "https://github.com/cnlohr/rv003usb"
Cloning into 'rv003usb'...

C:\test>cd rv003usb

C:\test\rv003usb>git submodule update --init
Submodule 'ch32v003fun' (https://github.com/cnlohr/ch32v003fun) registered for path 'ch32v003fun'
Cloning into 'C:/test/rv003usb/ch32v003fun'...
Submodule path 'ch32v003fun': checked out 'f1820f2995e146ce8be0e3a94cb34a53d8238061'

C:\test\rv003usb>cd bootloader

C:\test\rv003usb\bootloader>make_win.bat

C:\test\rv003usb\bootloader>REM Bad USB Operation on pgm-wch-linke.c ? Zadig!

C:\test\rv003usb\bootloader>del bootloader.bin
C:\test\rv003usb\bootloader\bootloader.bin konnte nicht gefunden werden

C:\test\rv003usb\bootloader>del bootloader.elf
C:\test\rv003usb\bootloader\bootloader.elf konnte nicht gefunden werden

C:\test\rv003usb\bootloader>del bootloader.hex
C:\test\rv003usb\bootloader\bootloader.hex konnte nicht gefunden werden

C:\test\rv003usb\bootloader>del bootloader.lst
C:\test\rv003usb\bootloader\bootloader.lst konnte nicht gefunden werden

C:\test\rv003usb\bootloader>del bootloader.map
C:\test\rv003usb\bootloader\bootloader.map konnte nicht gefunden werden

C:\test\rv003usb\bootloader>make build
riscv64-unknown-elf-gcc -o bootloader.elf bootloader.c ../rv003usb/rv003usb.S -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections -L../ch32v003fun/ch32v003fun/../misc -lgcc
riscv64-unknown-elf-size bootloader.elf
   text    data     bss     dec     hex filename
   1876       0      92    1968     7b0 bootloader.elf
riscv64-unknown-elf-objdump -S bootloader.elf > bootloader.lst
riscv64-unknown-elf-objdump -t bootloader.elf > bootloader.map
riscv64-unknown-elf-objcopy -O binary bootloader.elf bootloader.bin
riscv64-unknown-elf-objcopy -O ihex bootloader.elf bootloader.hex

C:\test\rv003usb\bootloader>..\ch32v003fun\minichlink\minichlink.exe -a -w bootloader.bin bootloader -B
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
```

Will try linux shortly

xsrf, 2024-02-27:
Also compiles on Linux. But it's actually 8 bytes bigger on linux 🤔
```
andreas@debian:~/Projects/tmp$ git clone "https://github.com/cnlohr/rv003usb"
Klone nach 'rv003usb'...
andreas@debian:~/Projects/tmp$ cd rv003usb/
andreas@debian:~/Projects/tmp/rv003usb$ git submodule update --init
Submodul 'ch32v003fun' (https://github.com/cnlohr/ch32v003fun) für Pfad 'ch32v003fun' in die Konfiguration eingetragen.
Klone nach '/home/andreas/Projects/tmp/rv003usb/ch32v003fun'...
Submodul-Pfad 'ch32v003fun': 'f1820f2995e146ce8be0e3a94cb34a53d8238061' ausgecheckt
andreas@debian:~/Projects/tmp/rv003usb$ cd bootloader/
andreas@debian:~/Projects/tmp/rv003usb/bootloader$ make clean
rm -rf bootloader.elf bootloader.bin bootloader.hex bootloader.lst bootloader.map bootloader.hex || true
andreas@debian:~/Projects/tmp/rv003usb/bootloader$ make
riscv64-unknown-elf-gcc -o bootloader.elf bootloader.c ../rv003usb/rv003usb.S -g -Os -flto -ffunction-sections -static-libgcc -march=rv32ec -mabi=ilp32e -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections -L../ch32v003fun/ch32v003fun/../misc -lgcc
riscv64-unknown-elf-size bootloader.elf
   text	   data	    bss	    dec	    hex	filename
   1884	      0	     92	   1976	    7b8	bootloader.elf
riscv64-unknown-elf-objdump -S bootloader.elf > bootloader.lst
riscv64-unknown-elf-objdump -t bootloader.elf > bootloader.map
riscv64-unknown-elf-objcopy -O binary bootloader.elf bootloader.bin
riscv64-unknown-elf-objcopy -O ihex bootloader.elf bootloader.hex
make -C ../ch32v003fun/minichlink all
make[1]: Verzeichnis „/home/andreas/Projects/tmp/rv003usb/ch32v003fun/minichlink“ wird betreten
gcc -o minichlink minichlink.c pgm-wch-linke.c pgm-esp32s2-ch32xx.c nhc-link042.c ardulink.c serial_dev.c pgm-b003fun.c minichgdb.c -lpthread -lusb-1.0 -ludev -O0 -g3 -Wall -I. 
gcc -o minichlink.so minichlink.c pgm-wch-linke.c pgm-esp32s2-ch32xx.c nhc-link042.c ardulink.c serial_dev.c pgm-b003fun.c minichgdb.c -lpthread -lusb-1.0 -ludev -O0 -g3 -Wall -I.  -shared -fPIC
make[1]: Verzeichnis „/home/andreas/Projects/tmp/rv003usb/ch32v003fun/minichlink“ wird verlassen
../ch32v003fun/ch32v003fun/../minichlink/minichlink -a -w bootloader.bin bootloader -B
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../ch32v003fun/ch32v003fun/ch32v003fun.mk:66: cv_flash] Fehler 224
andreas@debian:~/Projects/tmp/rv003usb/bootloader$ cd ..
andreas@debian:~/Projects/tmp/rv003usb$ cd ch32v003fun/
andreas@debian:~/Projects/tmp/rv003usb/ch32v003fun$ git checkout master
Vorherige Position von HEAD war f1820f2 Merge pull request #153 from alexhorner/spi_max7219
Zu Branch 'master' gewechselt
Ihr Branch ist auf demselben Stand wie 'origin/master'.
andreas@debian:~/Projects/tmp/rv003usb/ch32v003fun$ cd ..
andreas@debian:~/Projects/tmp/rv003usb$ git status
Auf Branch master
Ihr Branch ist auf demselben Stand wie 'origin/master'.

Änderungen, die nicht zum Commit vorgemerkt sind:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)
  (benutzen Sie "git restore <Datei>...", um die Änderungen im Arbeitsverzeichnis zu verwerfen)
	geändert:       ch32v003fun (neue Commits)

Unversionierte Dateien:
  (benutzen Sie "git add <Datei>...", um die Änderungen zum Commit vorzumerken)
	bootloader/bootloader.bin
	bootloader/bootloader.elf
	bootloader/bootloader.hex
	bootloader/bootloader.lst
	bootloader/bootloader.map

keine Änderungen zum Commit vorgemerkt (benutzen Sie "git add" und/oder "git commit -a")
andreas@debian:~/Projects/tmp/rv003usb$ cd bootloader/
andreas@debian:~/Projects/tmp/rv003usb/bootloader$ make clean
rm -rf bootloader.elf bootloader.bin bootloader.hex bootloader.lst bootloader.map bootloader.hex || true
andreas@debian:~/Projects/tmp/rv003usb/bootloader$ make
riscv64-unknown-elf-gcc -o bootloader.elf  bootloader.c ../rv003usb/rv003usb.S  -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections
riscv64-unknown-elf-size bootloader.elf
   text	   data	    bss	    dec	    hex	filename
   1884	      0	     92	   1976	    7b8	bootloader.elf
riscv64-unknown-elf-objdump -S bootloader.elf > bootloader.lst
riscv64-unknown-elf-objdump -t bootloader.elf > bootloader.map
riscv64-unknown-elf-objcopy -O binary bootloader.elf bootloader.bin
riscv64-unknown-elf-objcopy -O ihex bootloader.elf bootloader.hex
make -C ../ch32v003fun/minichlink all
make[1]: Verzeichnis „/home/andreas/Projects/tmp/rv003usb/ch32v003fun/minichlink“ wird betreten
gcc -o minichlink minichlink.c pgm-wch-linke.c pgm-esp32s2-ch32xx.c nhc-link042.c ardulink.c serial_dev.c pgm-b003fun.c minichgdb.c -lpthread -lusb-1.0 -ludev -O0 -g3 -Wall -DCH32V003 -I. 
gcc -o minichlink.so minichlink.c pgm-wch-linke.c pgm-esp32s2-ch32xx.c nhc-link042.c ardulink.c serial_dev.c pgm-b003fun.c minichgdb.c -lpthread -lusb-1.0 -ludev -O0 -g3 -Wall -DCH32V003 -I.  -shared -fPIC
make[1]: Verzeichnis „/home/andreas/Projects/tmp/rv003usb/ch32v003fun/minichlink“ wird verlassen
../ch32v003fun/ch32v003fun/../minichlink/minichlink -a -w bootloader.bin bootloader -B
Error: Could not initialize any supported programmers
Error: Could not initialize any supported programmers
make: *** [../ch32v003fun/ch32v003fun/ch32v003fun.mk:185: cv_flash] Fehler 224
andreas@debian:~/Projects/tmp/rv003usb/bootloader$ 
```

xsrf, 2024-02-27:
> i have installed GCC10, but I don't understand what the %path% step you mean is. now I see an error like this.

run `echo %path%` in your terminal to see the all directories your system is looking for executables. For me it contains `C:\SysGCC\risc-v\bin`.
run `where riscv64-unknown-elf-gcc` to see the full path of the executable found. For me it is `C:\SysGCC\risc-v\bin\riscv64-unknown-elf-gcc.exe`

cnlohr, 2024-02-27:
@xsrf do you have time to investigate?  I can believe something in the massive upheval broke when we added support for all the other chips.

anwfeb, 2024-02-28:
> > i have installed GCC10, but I don't understand what the %path% step you mean is. now I see an error like this.
> 
> run `echo %path%` in your terminal to see all directories your system is looking for executables in. For me it contains `C:\SysGCC\risc-v\bin`. run `where riscv64-unknown-elf-gcc` to see the full path of the executable found. For me it is `C:\SysGCC\risc-v\bin\riscv64-unknown-elf-gcc.exe`

Thank you for giving me the steps, and i was successful in doing that.
[image]

Later I will also try for Mac.

anwfeb, 2024-02-28:
what drivers can minichlink accept? the default driver after flashing the bootloader is "USB Input Device" I have tried changing with Zadig to WinUSB or libusbK or libusb-win32 but it's worse, minichlink doesn't detect it.

[image]

anwfeb, 2024-02-28:
then i tried to run flashing, but it showed error make: *** [busbflash] Error 2

[image]

xsrf, 2024-02-28:
> then i tried to run flashing, but it showed error make: *** [busbflash] Error 2

Don't use the testblink. I don't know what it is and I don't know what busbflash is.

Please try `demo_composite_hid`. It is tested on Windows, Linux and Mac and should work.

Btw: You need Zadig to change the driver of the Link-E flasher to be able to flash using minichlink via Link-E.
You do NOT need Zadig to flash software via the USB bootloader. The USB Bootloader enumerates as HID Input Device and that's fine. Minichlink can work with that. Do not change the driver for the bootloader.

xsrf, 2024-02-28:
> @xsrf do you have time to investigate? I can believe something in the massive upheval broke when we added support for all the other chips.

Not sure what I should investigate. For me it works on Linux and Windows with the old pinned ch32v003fun and with the latest one. I'm also not confident at all with all the headers and make files in different places.

cnlohr, 2024-02-29:
I believe busbflash was a test to put code into the scratchpad and execute it.  It makes me wonder if the USB device is not identified.  It would have to be flashed with the bootloader.

@anwfeb did you flash the bootloader?

LimesKey, 2024-04-21:
Got it compiled on Windows... but it took me a half hour to figure out that I needed to copy the contents of ch32v003fun into the ch32v003fun folder in rv003usb, how was I supposed to know to do this?

cnlohr, 2024-04-22:
> Got it compiled on Windows... but it took me a half hour to figure out that I needed to copy the contents of ch32v003fun into the ch32v003fun folder in rv003usb, how was I supposed to know to do this?

No, no.  You should be using git submodules.

When checking out github repos, use --recursive

If you didn't use --recursive, you can,
```
git submodule update --init --recursive
```

mothdotmonster, 2024-10-13:
I'm also not seemingly able to get anything to build on my machine, running Arch Linux.

The error I've got most is something like this:

```
riscv64-linux-gnu-gcc -o bootloader.elf  bootloader.c ../rv003usb/rv003usb.S  -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -Wl,--print-memory-usage -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections
In file included from /usr/riscv64-linux-gnu/usr/include/features.h:535,
                 from /usr/riscv64-linux-gnu/usr/include/bits/libc-header-start.h:33,
                 from /usr/riscv64-linux-gnu/usr/include/stdint.h:26,
                 from /usr/lib/gcc/riscv64-linux-gnu/14.2.0/include/stdint.h:9,
                 from ../ch32v003fun/ch32v003fun/ch32v003fun.h:223,
                 from bootloader.c:5:
/usr/riscv64-linux-gnu/usr/include/gnu/stubs.h:8:11: fatal error: gnu/stubs-ilp32.h: No such file or directory
    8 | # include <gnu/stubs-ilp32.h>
      |           ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [../ch32v003fun/ch32v003fun/ch32v003fun.mk:228: bootloader.elf] Error 1
```

I did manage to get the gamepad demo working at least, so I know my hardware is all good at least.

prosper00, 2024-10-13:
it looks like it's trying to pull in libc... i think you may be using the wrong GCC. Do you have riscv64-unknown-... or riscv64-none-... available?

mothdotmonster, 2024-10-22:
> it looks like it's trying to pull in glibc... i think you may be using the wrong GCC. Do you have riscv64-unknown-... or riscv64-none-... available?

I managed to get the compiling to seemingly work for at least some things by following the advice of https://github.com/cnlohr/ch32v003fun/issues/378#issuecomment-2312106663

The gamepad example works perfectly (which is admittedly a great start) but unfortunately I still can't figure out how the bootloader is supposed to work at all.

cnlohr, 2024-12-04:
Closing now, any more issues should be a new issue.

closed 2024-12-04 by cnlohr