cnlohr/rv003usb #50: does it need a quartz?

thirstyone, 2024-03-08:
It is absent on your schematic. Will it work without a crystal? Or I shouldn't even try? In general, will it work on a ch32v003 in sop8 package?

xsrf, 2024-03-08:
There is a schematic ;)
https://github.com/cnlohr/rv003usb?tab=readme-ov-file#it-requires-very-little-hardware

You do not need a crystal and it works with the SOP8. I use this configuration on two of my boards with the CH32V003J4M6:
```
PC1 D+
PC2 D-
PC4 D-_PU
```

thirstyone, 2024-03-08:
Yep, I've seen the schematic. It's incredible that it doesn't need an external crystal! Thank you!

cnlohr, 2024-03-09:
Just FYI - the reason it doesn't need a crystal is because it tunes the internal oscillator based on the SOFs (or SE0s)

closed 2024-03-09 by cnlohr

doragasu, 2024-05-04:
> Just FYI - the reason it doesn't need a crystal is because it tunes the internal oscillator based on the SOFs (or SE0s)

Oh, that could explain why it is **not** working when I enable HSE, maybe HSI is a requirement?

cnlohr, 2024-05-07:
It is possible the HSI might be a requirement, since it uses the SOFs to retime the HSI.