cnlohr/rv003usb #55: PORT A for D+ D- Possible?

ATP-IOT, 2024-05-01:
It's possible to use port A For the D+ and D- lines?

ATP-IOT, 2024-05-01:
#define USB_PORT A     // [A,C,D] GPIO Port to use with D+, D- and DPU
#define USB_PIN_DP 1   // [0-4] GPIO Number for USB D+ Pin
#define USB_PIN_DM 2   // [0-4] GPIO Number for USB D- Pin
//#define USB_PIN_DPU 5  // [0-7] GPIO for feeding the 1.5k Pull-Up on USB D- Pin; Comment out if not used / tied to 3V3!

cnlohr, 2024-05-02:
You got it working, right?  Close issue?

NoNamedCat, 2024-05-02:
No, the device is not recognized by the system. I put a 1.5K resistor on the D- line but the OS throw me an error when I connect the device.

cnlohr, 2024-05-02:
Do you have a logic analyzer you could use here to see what could be going on.  There's lots of possibilities that would cause issues, that may be totally unrelated.  I know other folks have a configuration like yours on port A.

NoNamedCat, 2024-05-03:
I will try to make some experiments (change the GPIOs, the cable, the IC....) If nothing else work, I will use a digital logic analyzer. I don't have one, but i have access to one of those.
Tks a Lohr!

cnlohr, 2024-05-03:
Totes recommend b-lining it to the scope capture so we can peek!  

Can you draw a full schematic?

NoNamedCat, 2024-05-03:
[image]
[image]
[image]

xsrf, 2024-05-03:
@NoNamedCat your USB signals are swapped on the USB connector. Your 1.5k pull-up is connected to D+ 😬

NoNamedCat, 2024-05-03:
Tks! I will do the changes, and try again. Close this

xsrf, 2024-05-03:
Now the pull-up is wrong - still connected to D+. The pull-up needs to be connected to D-/PA1

NoNamedCat, 2024-05-09:
[image]

closed 2024-05-14 by ATP-IOT