cnlohr/rv003usb #61: Question - Could this be used as a programmer for ch32v003fun?

Willyarma, 2024-07-02:
I was wondering, could this USB code be adapted to be used as a programmer for ch32v003fun (for programming another device)?
I know it would be a catch 22 situation, except since I've just bought 100 of these, I could make programmers for my friends.
I have some ESP32S2s so not an issue for me.

cnlohr, 2024-07-03:
It totally could.  It's a bit tricky because both it and the USB subsystem requires very low latency control timing, but it is possible to do with DMA. It would be a fun project to get going.  But it would take learning a little bit about chaining GPIO DMA off of timers. I would love to help someone seek this if they wanted.

Willyarma, 2024-07-04:
Thanks for your reply.
The chips haven't arrived and I have no experience with these yet.
I'd have a look myself if it were simple enough, but maybe its something someone else might be interested in?

cnlohr, 2024-07-04:
It will take some effort, and a logic analyzer, but it should be tractable, I have a reference implementation here - you would need to figure out how to send `1`'s and a `0`'s reliably, as well as reading bits. https://github.com/cnlohr/esp32s2-cookbook/blob/master/ch32v003programmer/main/ch32v003_swio.h#L94-L182

closed 2024-09-02 by Willyarma