cnlohr/rv003usb #63: compile demo_hidapi project with error

yefeng0, 2024-09-23:
compile demo_hidapi project with error, here is log below:
`
$ make
riscv64-unknown-elf-gcc -E -P -x c -DTARGET_MCU=CH32V003 -DMCU_PACKAGE= -DTARGET_MCU_LD=0 -DTARGET_MCU_MEMORY_SPLIT= ../ch32v003fun/ch32v003fun/ch32v003fun.ld > ../ch32v003fun/ch32v003fun/generated_ch32v003.ld
riscv64-unknown-elf-gcc -o demo_hidapi.elf ../ch32v003fun/ch32v003fun/ch32v003fun.c demo_hidapi.c ../rv003usb/rv003usb.S ../rv003usb/rv003usb.c  -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I../lib -I../rv003usb -Wl,--print-memory-usage -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ../ch32v003fun/ch32v003fun/generated_ch32v003.ld -Wl,--gc-sections
../ch32v003fun/ch32v003fun/ch32v003fun.h: Assembler messages:
../ch32v003fun/ch32v003fun/ch32v003fun.h:13212: Error: unrecognized opcode `csrr a5,0x804', extension `zicsr' required
../ch32v003fun/ch32v003fun/ch32v003fun.h:13218: Error: unrecognized opcode `csrw 0x804,a5', extension `zicsr' required
lto-wrapper.exe: fatal error: D:\msys64\ucrt64\bin\riscv64-unknown-elf-gcc.exe returned 1 exit status
compilation terminated.
D:/msys64/ucrt64/bin/../lib/gcc/riscv64-unknown-elf/14.2.0/../../../../riscv64-unknown-elf/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
make: *** [../ch32v003fun/ch32v003fun/ch32v003fun.mk:228：demo_hidapi.elf] 错误 1
`

maxgerhardt, 2024-10-01:
Then there's a `.option arch, +zicsr` missing somewhere, check out https://github.com/maxgerhardt/ch32v003-vanilla-gcc.

cnlohr, 2024-10-01:
I believe I fixed this upstream, can you update, then update ch32v003fun to master, and see if your issue persists?