cnlohr/rv003usb #67: Linked GCC10 Toolchain is flagged as Virus/potentially dangerous

cad435, 2024-11-04:
When trying to download the prebuild GCC10 Toolchain for RISCV from the provided link:

https://gnutoolchains.com/risc-v/

Virustotal freaks out a bit:
https://www.virustotal.com/gui/file/9fdb75f4c67a6975f362ae61116c65bef5ba1e841e2f22a3b14d2da9df0e07fa

Is it safe to use?

 - cad435

cnlohr, 2024-11-06:
I have not seen this!  We will discuss on my Discord.  Thank you for reporting it.

cad435, 2024-11-06:
You're welcome! Thank you for looking in

I just want to add that it actually is very easy to make a Ubuntu WSL, install the required packages and compile the bootloader this way. (Never done this before, defintely will use it more often in the future)

for me, with (I'd say) beginner Linux skills definately less effort than installing a toolchain on windows...

cnlohr, 2024-11-06:
It is unacceptable for users to install WSL to use this tool.  WSL requires >100MB of space, which is out of the question for supporting embedded development for a platform... not to mention additional services will need to run.

The link above is a single installer that includes all needed components, so the complete installation steps are:

1. Use the GCC installer from the above link.
2. Get a copy of ch32v003fun.
3. Run "make" in the example of your choice from command-line.

Anything more complicated than that is not going to be tolerable to our audience. 

-- on the reporting end --

So far, one other user is reporting that there is no issue.  We are continuing to investigate.

Charles

cad435, 2024-11-06:
Hmm, yeah I definately can see that points in regards to WSL...

cnlohr, 2024-11-07:
It appears that this _might_ be a false alarm.  It is possible that the install methodology of this GCC compiler toolchain is similar to an attack vector of muddywater.  But, still looking.  I've reached out to sysprogs to discuss with them.

cnlohr, 2024-12-04:
After talking to others, I think this is a false alarm, but we currently have an issue open with xpack, to try to get them to include ilp32e, rv32ec https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/issues/38

cnlohr, 2024-12-04:
xpack binaries are compatible rv003usb.  there is a new build coming out with explicit ilp32e--c support, which is even better. I will migrate from sysprogs to xpack.