cnlohr/rv003usb #72: CDC USB on Windows

cuongpham2110, 2024-12-12:
Hi, 
I have problem in Windows that USB device can be recognized but have exclamation mark on device manager. The error code is 10 (cannot start). Is that your error you are facing now? I've tested CDC USB on linux and it worked fine with /dev/ttyACM0. Do you have any ideas for fixing this bug?
Thanks

cnlohr, 2024-12-12:
If you have some time and can poke at it it would be greatly appreciated.  I believe I only ever got it working reliably in Linux.  but I don't really remember.  That's why it's in the testing/ folder.

cnlohr, 2024-12-13:
My issue is that if I define something other than interrupt, Linux will obliterate the endpoint. We are currently chatting in-discord about this.

cnlohr, 2024-12-13:
> Hi, I find that you define `interrupt` Attribute for all Endpoint. I think it may violate some rules in `usbser` and leads to error code 10. I change `usb_config.h` to Bulk Endpoint IN/OUT (In CDC Data Interface) and keep Interrupt Endpoint of CDC Control Interface, the result is worse. USB device is even disconnect to host and Windows warning malfunction. So I want to know if the source code support Bulk Endpoint? Do I need focus on somewhere in the code ? Thanks

Inherently, the underlying stack is happy to talk bulk.  It's only the descriptors that explain the usage/access pattern from the host.  So if you can play around, maybe you can find a configuration that works?

cuongpham2110, 2024-12-14:
I am using CH32V203 CDC USBD configuration as reference. It has hard USB core and it works perfectly on Windows. Every USB config I use for CH32V003 is same as CH32V203 except IN/OUT Bulk endpoint. As I mention before, if change it from Interrupt to Bulk, USB device is malfunctioned immediately. 
I've upload working USB config (analyzed by USBlyzer) for CH32V203 here: [worked_ch32v203.txt](https://github.com/user-attachments/files/18133924/worked_ch32v203.txt)
Hope that file may help us to find a solution for this bug.

kimstik, 2024-12-19:
> My issue is that if I define something other than interrupt, Linux will obliterate the endpoint. We are currently chatting in-discord about this.

Yes, that's correct. According to the USB (Universal Serial Bus) specifications, low-speed devices do not support bulk endpoints.

cnlohr, 2024-12-20:
Unfortunately USBFS/HS examples don't really work for that.

My hope is that there's some way "around" it like there is with network CDC stuff, to allow you to use interrupt endpoints as bulk endpoints.

kimstik, 2024-12-26:
FYI, https://github.com/lemcu/LemcUSB/issues/1#issuecomment-38436738

By the way, they realized that low-speed USB has a small payload, so they receive it directly into 3 processor registers (12 bytes) without touching the memory.
neat and logical...

cnlohr, 2024-12-26:
Just because they couldn't do it doesn't mean it can't be done.  But, yes, it is totally do-able, but, it's less convenient to work with, and makes the code bigger to do it that way.