cnlohr/rv003usb #74: Bootloader doesn't meet size window when compiled

Plaque-fcc, 2024-12-22:
```
$ make -C bootloader build
make: Entering directory '/build/rv003usb/bootloader'
riscv64-linux-gnu-gcc -o bootloader.elf  bootloader.c ../rv003usb/rv003usb.S  -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003=1 -static-libgcc -I/usr/include/newlib -I../ch32v003fun/ch32v003fun/../extralibs -I../ch32v003fun/ch32v003fun -nostdlib -I. -Wall -I. -I../lib -DUSE_TINY_BOOT -I../rv003usb -Wl,--print-memory-usage -L../ch32v003fun/ch32v003fun/../misc -lgcc -T ch32v003fun-usb-bootloader.ld -Wl,--gc-sections
/tmp/ccwYoG6F.s: Assembler messages:
/tmp/ccwYoG6F.s:561: Warning: setting incorrect section attributes for .rodata
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: bootloader.elf section `.text' will not fit in region `FLASH'
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: region `FLASH' overflowed by 720 bytes
/usr/lib/gcc-cross/riscv64-linux-gnu/11/../../../../riscv64-linux-gnu/bin/ld: bootloader.elf: error: PHDR segment not covered by LOAD segment
Memory region         Used Size  Region Size  %age Used
           FLASH:        2640 B       1920 B    137.50%
             RAM:         280 B         2 KB     13.67%
collect2: error: ld returned 1 exit status
make: *** [../ch32v003fun/ch32v003fun/ch32v003fun.mk:235: bootloader.elf] Error 1
make: Leaving directory '/build/rv003usb/bootloader'
```

I just cloned the project locally (with submodule), and it outputs this message that the bootloader is oversized.

Why is this happening and how can we fix it?

cnlohr, 2024-12-22:
Please run `riscv64-linux-gnu-gcc -v` -- for the bootloader you should be using gcc14

Plaque-fcc, 2024-12-22:
> Please run `riscv64-linux-gnu-gcc -v` -- for the bootloader you should be using gcc14

gcc version 11.4.0

Feels like I'm stuck at GCC v12 for now. Thanks for your quick reply (you can probably add minimum requirements in the README, btw).

Plaque-fcc, 2024-12-22:
Or even with GCC v11 for RISCV64.

cnlohr, 2024-12-27:
Ok, honestly, gcc v11 _should_ be OK.  What configuration do you have?  Some configurations for the bootloader are not able to fit in the bootloader space.  Or is your install a virgin install?

Plaque-fcc, 2025-01-02:
> Ok, honestly, gcc v11 _should_ be OK. What configuration do you have? Some configurations for the bootloader are not able to fit in the bootloader space. Or is your install a virgin install?

Totally clean install. I may be getting something wrong. Tried with Ubuntu 22.04.5 LTS x86_64. Will likely upgrade it to the next LTS in a few days.

I didn't configure anything in the bootloader at all. Fresh clone, doesn't work → came to ask.

Plaque-fcc, 2025-01-04:
Anyway, my (other) ARM64 machine can't compile it (guess there are difficulties cross-compiling for RISC-V on anything than AMD64 and probably x86 so far unless you're on bleeding edge or building toolchain yourself).

Therefore, I'm going to upgrade my old LTS instance to the latest LTS one and give it another try. Will get back with the news.

Plaque-fcc, 2025-01-15:
Good news, I wasn't using the right toolchain, solved with building one from source [#493](https://github.com/cnlohr/ch32v003fun/issues/493).

Thanks for your help, closing this one.

closed 2025-01-15 by Plaque-fcc

Plaque-fcc, 2025-02-11:
Branch `master`, commit dd156840f9a9f54dd2aa4e9d8dc3bd7e688b8963, `riscv64-unknown-elf-gcc` version 10.2.0.

Default bootloader config:
```
Memory region         Used Size  Region Size  %age Used
           FLASH:        1892 B       1920 B     98.54%
             RAM:          96 B         2 KB      4.69%
```

Everything is just fine, thank you!

cnlohr, 2025-02-11:
Woot! Glad it's all good.