cnlohr/rv003usb #76: Not working on Windows, problem with descriptor failure

luisf18, 2025-01-06:
Hello, I'm trying to run the example "demo_gamepad" but it's not working for me :/. Windows says that the descriptor request failed.

*"Windows has stopped this device because it has encountered problems. (Code 43)
A USB device descriptor request failed."*

[image]
[image]

Has anyone had this problem? What can I do?

**Pinout using ch32v003**
PC1 -> D-
PC2 -> D+
without DPU

cnlohr, 2025-01-07:
There are a lot of possible issues.  But, unfortunately that message doesn't mean anything, it does not indicate any operation that I'm aware of.

Please include more information, 
1. Do you have a 1.5k ohm pull up on D- installed?
2. Can you probe voltage on D- and D+?
3. Please include your source code and/or changes, since I don't think we use PC1/PC2 for any of the examples in this repo.

luisf18, 2025-01-08:
Hello! Yes, I connected the pull up resistor to D- and changed the port to C (in the example it was A). Then I recompiled and uploaded the code. I will try to measure the signals on the D+ and D- pins and send them here.

luisf18, 2025-01-08:
I used a logic analyzer to generate this file. Is it enough? I’m not used to sharing this type of measurement, so if there’s a better format than CSV, please let me know.
[ch32v003_USB.csv](https://github.com/user-attachments/files/18341462/ch32v003_USB.csv)

luisf18, 2025-01-13:
Unfortunately, I keep trying, but nothing seems to work. I tried uploading the examples without changing anything, using pins PD3, PD4, and PD5, but it didn't work :/. Could you answer a few questions to help me debug what's happening?

1 - When the CH32V003 starts up, does it send any signal on the D+ and D- pins when no USB is connected?
2 - I don’t know much about the USB protocol. How can I use a logic analyzer to check what’s happening and whether the CH32V003 is at least trying to communicate?
3 - I tried connecting it to my android phone using an OTG cable, but it wasn’t recognized either. Should it work in this case?

ZUNA8415, 2026-09-06:
I also ran into the same problem, and I’m writing this because I think I’ve found a solution. (Please understand that my English is not very good.)

To get straight to the point, I flashed it using the **-w bootloader.bin 0x08000000** command directly instead of **-w bootloader.bin bootloader**, and it worked well as follows.

```
minichlink.exe -a -w bootloader.bin 0x08000000 -B -c COM8
minichlink version - 38e653f8354ea8fc19da5f2595cf9958d26738e7
VID:0x1209, PID:0xb003
Opening serial port COM8 at 115200 baud.
Ardulink: synced.
Found Ardulink Programmer
Ardulink: target power 1
Interface Setup
Detected CH32V003
Flash Storage: 16 kB
Part UUID: b9-0a-ab-cd-23-4b-be-74
Part Type: 00-30-05-10
Read protection: disabled
Detecting Memory Area
Writing image

Image written.
```

```
minichlink.exe -i
minichlink version - 38e653f8354ea8fc19da5f2595cf9958d26738e7
VID:0x1209, PID:0xb003
Found B003Fun Bootloader
Halting Boot Countdown
Detected CH32V003
HID buffer: 1152 bytes
Flash Storage: 16 kB
Part UUID: b9-0a-ab-cd-23-4b-be-74
Part Type: 00-30-05-10
Read protection: disabled
Interface Setup
USER/RDPR  : e817/5aa5
DATA1/DATA0: ff00/ff00
WRPR1/WRPR0: 00ff/00ff
WRPR3/WRPR2: 00ff/00ff
R32_ESIG_UNIID1: b90aabcd
R32_ESIG_UNIID2: 234bbe74
R32_ESIG_UNIID3: ffffffff
```

Below is the process of the journey to solving this problem, so it is a long text and should only be read by those who are interested.

First, I used a breadboard for testing before making the PCB, and the pull-up resistor for D-, I used a 1KΩ resistor (the guide recommends 1.5KΩ, but I thought 1K to 2K would work fine, so I used the 1KΩ resistor that was commonly found right next to me.). And for stabilization, I connected 47Ω resistors in series on each data line. (This is optional, and I have confirmed that it works fine even without it.)

The configuration in usb_config.h was in default state without being modified as follows, (When binding a pull-up to a VCC, it just works very well even if you don’t commented out so don’t worry too much.)

```
#define USB_PORT D     // [A,C,D] GPIO Port to use with D+, D- and DPU
#define USB_PIN_DP 3   // [0-4] GPIO Number for USB D+ Pin
#define USB_PIN_DM 4   // [0-4] GPIO Number for USB D- Pin
//#define USB_DPU_PORT A  // [A,C,D] Override GPIO Port for DPU
#define USB_PIN_DPU 5  // [0-7] GPIO for feeding the 1.5k Pull-Up on USB D- Pin; Comment out if not used / tied to 3V3!
```

When I compiled and flashed the bootloader, it was in the same "descriptor failure" state as described. Specifically, the board silk I use did not have PD5 printed on it, and after looking into it, I found that the UART TX pin is PD5, and although I connected it to the D- pin, there was no response at all. So I try to connect the pull-up directly to VCC, there was a response but it was in the "descriptor failure" state.

I thought about giving up, but decided to at least upload an examples, so I compiled **demo_composite_hid** and uploaded it by **-w demo_composite_hid.bin flash**, and it was recognized as an HID device, my mouse cursor began moving around drawing small squares!

I changed the macro for defining the timeout in **bootloader.c** to **#define BOOTLOADER_TIMEOUT_PWR 0**, compiled it, and uploaded it, but for some reason my mouse kept drawing squares.

Even after wiping the entire chip (-E) and flashing, it was the same "descriptor failure" state, and since it was strange that it started in the firmware area rather than the bootloader, so I decided to try writing directly to the starting address **0x08000000**, and after flashing it that way, it finally started being recognized properly!

I need to see how the **-w bootloader.bin bootloader** actually works to know for sure, but anyway it might be being written in a strange location for some reason.

[... 62 lines cut ...]
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff806, ff00 != f806
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff808, 00ff != f808
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff80a, 00ff != f80a
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff80c, 00ff != f80c
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultWriteHalfWord
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
Fault on DefaultReadHalfWord
Warning when writing option bytes at 1ffff80e, 00ff != f80e
Whole-chip erase
Fault on op (DMABSTRACTS = 08000402) (100) (Processor not halted.) DMSTATUS: 004c0c82
```

ZUNA8415, 2026-09-08:
I think I may have also discovered the cause of the chip failure.

Similarly to get straight to the point, it seems the problem was not Ardulink’s fault, but rather an issue may caused by the **clock settings**.

OpenWCH's https://github.com/openwch/arduino_core_ch32 board manager (for convenience, call it as **"OpenWCH core"** from now), it is a core for EVT boards, and the CH32V003 EVT board has an external xtal.

Perhaps for that reason, **system/CH32V00x/USER/system_ch32v00x.c** says using clock source from external xtal as follows.

```
/* 
* Uncomment the line corresponding to the desired System clock (SYSCLK) frequency (after 
* reset the HSI is used as SYSCLK source).
* If none of the define below is enabled, the HSI is used as System clock source. 
*/

//#define SYSCLK_FREQ_8MHz_HSI    8000000
//#define SYSCLK_FREQ_24MHZ_HSI   HSI_VALUE
//#define SYSCLK_FREQ_48MHZ_HSI   48000000
//#define SYSCLK_FREQ_8MHz_HSE    8000000
//#define SYSCLK_FREQ_24MHz_HSE   HSE_VALUE
#define SYSCLK_FREQ_48MHz_HSE   48000000
```

In the latest **v1.0.4** released package, that can be downloaded by adding a json link, you can see that,

https://github.com/openwch/arduino_core_ch32/blob/1.0.4/system/CH32V00x/USER/system_ch32v00x.c

But for some reason I’m not sure, on the latest version of the repository, it appears to be commented out.

https://github.com/openwch/arduino_core_ch32/blob/main/system/CH32V00x/USER/system_ch32v00x.c

```
//#define SYSCLK_FREQ_8MHz_HSI    8000000
// #define SYSCLK_FREQ_24MHZ_HSI   HSI_VALUE
// #define SYSCLK_FREQ_48MHZ_HSI   48000000
//#define SYSCLK_FREQ_8MHz_HSE    8000000
// #define SYSCLK_FREQ_24MHz_HSE   HSE_VALUE
// #define SYSCLK_FREQ_48MHz_HSE   48000000
```

On the other hand, the board manager at https://github.com/AlexanderMandera/arduino-wch32v003 based on the ch32fun core (for convenience, call it as **"ch32fun core"** from now), clearly set for using the clock from internal source.

From **cores/arduino/Arduino.h**

```
#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK
#define SYSTICK_USE_HCLK
```

From **cores/arduino/wiring.c**

```
void init( void )
{
	SystemInit48HSI();

	SysTick->CTLR = 0;
	NVIC_EnableIRQ(SysTicK_IRQn);

	SysTick->CMP = DELAY_MS_TIME - 1;
	SysTick->CNT = 0;
	systick_counter = 0;

	SysTick->CTLR = SYSTICK_CTLR_STE | SYSTICK_CTLR_STIE |
					SYSTICK_CTLR_STCLK;
}

#define SYSTEM_CORE_CLOCK 48000000
#define APB_CLOCK SYSTEM_CORE_CLOCK
[... 29 lines cut ...]
The code was so simple that I was confident there probably wouldn’t have been any problems, and no matter how many issues it might have caused due to lack of optimization, I didn’t think it was incapable of displaying 1fps environment.

After seeing the reset occur, I immediately tried reinforcing it using additional components such as capacitors and resistors, but there was no change.

In fact, after continuing to use it like that, the screen stopped showing anything at all, and when I checked, the MCU had died. (First death chip detected!)

This is a dead chip that has caused a short circuit (more precisely, a very low resistance) between the VCC and GND as mentioned above, and although the resistance increases over time, it drops again when the power is reapplied, causing the problem.

There were too many validations to dismiss it as simply a chip or board defect, or a soldering mistake, (I checked every pin with a multimeter to make sure there were no short circuits between pins legs and the soldering was done correctly), and since almost identical symptoms occurred on multiple chips, I initially suspected Ardulink.

However, even after trying to upload with WCH-LinkE, the same issue reproduced, and it even feels like it broke down even sooner. (Of course, it was very likely just a coincidence.) So I started to think that it probably wasn’t Ardulink’s fault, and when I thought about what was different from the beginning, I realized it was the difference in the board core selected in the Arduino IDE.

In **ch32fun core**, the LED was working fine, the WS2812 control was also working fine, and because the I2C compilation failed, I switched to **OpenWCH core**. However, under the naive assumption that there wouldn’t be major structural differences, I moved straight to the I2C test stage without going through the previous tests, so I decided to try running a Blink test compiled in the **OpenWCH core** environment.

Bring the new clean chip and flash it using WCH-LinkE. The moment I thought it seemed to be working well, I looked at the LED and saw it was slow. The default settings are ON for 500ms and OFF for 500ms, but the combined interval seems to be well over 1 second.

Since it felt off, I changed it to 250ms + 250ms, and only now did it seem like the interval was about 1 second. In other words, it operates at about 0.5 times the programmed value, and I thought the only possible cause for this problem was the clock. (An example such as the difference in time over a period of time when the external 16MHz and internal 8MHz clocks of the ATmega328P chip on Arduino Nano or UNO are set differently.)

I analyzed the cause by going over the basic concepts, such as the "built-in RC oscillator operating at 24MHz... and ×2 48MHz by PLL...", but I couldn’t identify it immediately. After investigating the board manager's settings and other details, I obtained the information above and reached my current conclusion.

[... cut ...]
The post I wrote earlier (about flashing the bootloader) and this post as well, sometimes something cannot be explained clearly, yet can be resolved by a specific method. I wrote about the process I went through with the hope that, for people facing the same or similar issues, this method would help if they read this and give it a try.

As a result, there may be a lot of incorrect information or non-expert knowledge. I’d appreciate your understanding kindly, and if there’s anything I can improve or if you have any advice, please feel free to share them. Thanks!

ZUNA8415, 2026-09-10:
After several experiments, I think I've finally found the real cause of the chip failure.

Similarly, to get straight to the point, it seems the cause was that the **adapter board’s traces were too thin**.

As I mentioned above, I performed many basic verifications multiple times. I think I tried every possible possible cases.

- [x] Check the wiring connection (between the IC’s leg and the hole on the adapter board)
- [x] Pin-to-pin short check (due to the very tight pin spacing of the TSSOP-20 package, check more thoroughly)
- [x] Decoupling capacitor between VCC and GND (the basics of stabilization)
- [x] Try to connect a 10K external pull-up to the NRST pin (there is already an internal pull-up, but make it a bit stronger)
- [x] Try to connect a capacitor between NRST and GND
- [x] Also try to connect a 10K pull-up to the SWIO pin

And... I’ll keep saying it, but far too many chips have been sacrificed to say it was just a simple soldering mistake or a chip or board defect, and even after multiple verifications, it always ended up dying after working to some extent at first, showing abnormal symptoms. (Reading the board’s UUID and register information, chip erase, unlock, upload bootloader or write code to flaah = everything works fine at first)

The Ardulink problem and the clock configuration issue that I had hypothesized were all rejected, and the chip quickly caused same problem.

I wondered whether the problem was because solder was used minimally to prevent short, but even after using plenty of solder with solder paste, the problem didn’t solved.

So while considering many possibilities, other pins may not have any issues, but maybe the **GND plane** is absent, so the **thin traces** may have caused a **comprehensive problem**.

So I thought about how to prove this hypothesis, and decided to test it on the **SOP package board surface**, which is about **twice the spacing of the SSOP**. It’s a really good moment because the board is double-sided.

<details><summary>Click to Expand</summary>
ㅤ

SSOP and TSSOP have a pin pitch of **0.65 mm**, while SOP is **1.27 mm (50 mil)**. Half of 1.27 mm is about 0.635 mm, so it roughly matches, but as the number of pins increases, small difference at each pins results in a cumulative error.

So, I cut the pins (**Alternating**, for example : leaving 1 = PD4, cutting 2 = PD5, leaving 3 = PD6, cutting 4 = PD7...), and since the essential **VCC, GND, and SWIO (PD1)** pins were left, I was able to set up a good environment for at least some basic testing.

[image]

By taking advantage of the cumulative error mentioned above, I left the PC0 and PC1 pins at the bottom usable without being cut off.

[image]

[image]

</details>

As a result, after building and testing it this way, it operated flawlessly for tens of minutes even **without the pull-up resistor and without even the decoupling capacitor**, and uploading was possible immediately without any issues whenever I wanted.

However, there is also a tutorial that uses the same adapter board, so this part is still unclear...

https://www.instructables.com/DIY-CH32V003-Development-Board-10-Cent-Devoard-for

While I can’t draw a definitive conclusion, what is clear is that, among all the configurations I’ve tested so far, using a wider pad with more solder is the **only one that has operated reliably without any issues**, and it may be an issue that only applies to me due to some particular environment.

However, it would be helpful if when someone tries to resolve a similar issue later, this information can be considered as a reference to fix the problem. Thank you.