cnlohr/rv003usb #78: Bootloader writes but doesn't enumerate

LimesKey, 2025-01-25:
Hi, I made a custom PCB for the `CH32V003F4U6` following the exact as in this repo's README, here's my schematic,

[image]
Here's a photo of my PCB for reference,
[image]

I used the `WCH-LinkE Mini` and connected the following,
| WCH-LinkE    | CH32V003 |
| -------- | ------- |
| GND  | GND    |
| SWC/TCK |  PC5/SCK    |
| SWD/TMS    | PD1/SWIO    |

And after a little trial and error, I got it to successfully flash,
```bash
~/Downloads/rv003usb-master/bootloader ❯ sudo make                                                10:17:59 PM
make -C ../ch32v003fun/minichlink all
make[1]: Entering directory '/home/ryan/Downloads/rv003usb-master/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ryan/Downloads/rv003usb-master/ch32v003fun/minichlink'
../ch32v003fun/ch32v003fun/../minichlink/minichlink -a -w bootloader.bin bootloader -B
Found WCH Link
WCH Programmer is LinkE version 2.10
Full Chip Type Reply: [8] 82-0d-05-09-00-31-05-10-03
Detected: CH32V003
Setup success
Flash Storage: 16 kB
Part UUID    : ad-86-ab-cd-15-e5-bc-92
PFlags       : ff-ff-ff-ff
Part Type (B): 00-00-00-12
Read protection: disabled
Interface Setup
Writing image

Image written.
```
I left alone the `usb_config.h` as I had the same pinouts as in the default README. I assume the other default settings would work fine. 

However, both on Linux Mint and Windows, I get no evidence of any USB device connected. I'm not sure what to blame, the PCB and likely the USB-C port, or the bootloader. I've already verified that the USB-C is delivering power and there aren't any shorts, but maybe there's still some issue?

What else should I try? I have yet to run any example code as frankly I don't know how.

tintinnabulate, 2025-02-16:
A couple of things to try: 

1. Try running the configurebootloader app in the subfolder. Note, you may need to first apply this patch: https://github.com/cnlohr/rv003usb/pull/83
2. Flash one of the demos to your board, e.g. demo_gamepad. I think the demos set up crucial usb descriptor info that is necessary for the usb protocol to work 
3. If those two fail, try taking a look at `dmesg` on Linux to see if the device is detected in any way, and use a logic analyzer connected to the DP/DM pins to check if there's any usb chatter starting up when you plug in your PCB.

cnlohr, 2025-04-21:
Did you get any further with this?  Should we close this issue?

One other note from me:
1. You only need to connect PD1.  
2. Can you share more of your code?