cnlohr/rv003usb #93: I have some problems trying to use the rvswdio_programmer

NoNamedCat, 2025-04-18:
When i try to use the mnichlinke.exe from the releases (v1.0rc2) I got this.

\minichlink\1.0.0> ./minichlink.exe -w ./fade.ino.bin flash -b
Error: Got error -1 when sending hid feature report.
Error: Got error -1 when sending hid feature report.
Error: Got error -1 when sending hid feature report.
Found ESP32S2-Style Programmer
Error: Got error -1 when sending hid feature report.
Error: Got error -1 when sending hid feature report.
Error: Could not read dmstatus.
Could not setup interface.

cnlohr, 2025-04-18:
You may need to install the winusb drivers.  Please see here: https://github.com/cnlohr/ch32fun/tree/master/misc/drivers_for_WCH-LinkE/R0-1v3 and comments here if that doesn't work about zadig. https://github.com/cnlohr/ch32fun/wiki/Installation#windows

NoNamedCat, 2025-04-25:
Tks a lot cnLohr. Will test the driver and comment this topic when i try it

zoobab, 2025-04-29:
Would it be possible to have some hardware schematics on this is supposed to work?

cnlohr, 2025-04-29:
They are posted in the rvswdio programmer folder, here : https://github.com/cnlohr/rv003usb/tree/master/rvswdio_programmer#a-minichlink-compatible-programmer

m039, 2025-08-22:
I encountered the same issue with PlatformIO, but when I use minichlink from the command line, it works. I’ve found that the root of the issue is that minichlink in PlatformIO and minichlink from the main repo (ch32fun) are different. I’ve replaced the malfunctional minichlink, and now everything is working right.