cnlohr/rv003usb #94: unable to flash testblink example to the USB-bootloader

wohltat, 2025-04-19:
Hi,

i've some problems to get the USB flashing to run, that was shown in the youtube video (https://www.youtube.com/watch?v=j-QazXghkLY).

What i've done so far:
- i made a board that works with the USB examples (keyboard, gamepad) when flashed via WCH-LinkE.
- i configured the bootloader according to the description and flashed the bootloader
- the usb device (idVendor=1209, idProduct=b003) shows up and stays (`#define BOOTLOADER_TIMEOUT_USB 0`)

When trying to flash the `testblink` example then i just get
```
./busbflash                                     
HD: (nil)                                       
make: *** [../../ch32v003fun/ch32fun/ch32fun.mk:292: cv_flash] Error 252 
```

Here is the complete output when i try to replicate the video:
```
> make clean all flash                       
rm -rf testblink.elf testblink.bin testblink.hex testblink.lst testblink.map testblink.hex ../../ch32v003fun/ch32fun//generated__.ld || true
rm -rf busbflash                                
riscv64-unknown-elf-gcc -o testblink.elf  testblink.c   -g -Os -flto -ffunction-sections -fdata-sections -fmessage-length=0 -msmall-data-limit=8 -march=rv32ec -mabi=ilp32e -DCH32V003 -static-libg
cc -I/usr/include/newlib -I../../ch32v003fun/ch32fun//../extralibs -I../../ch32v003fun/ch32fun/ -nostdlib -I. -Wall -I../rv003usb -I../lib -Wl,--print-memory-usage -Wl,-Map=testblink.map -L../../
ch32v003fun/ch32fun//../misc -lgcc -lgcc -T scratchpad.ld -Wl,--gc-sections
Memory region         Used Size  Region Size  %age Used                                                                                                                                            
      SCRATCHPAD:          40 B        128 B     31.25%                                                                                                                                            
          MEMORY:          0 GB        512 B      0.00%                                                                                                                                            
riscv64-unknown-elf-objdump -S testblink.elf > testblink.lst                                                                                                                                       
riscv64-unknown-elf-objcopy  -O binary testblink.elf testblink.bin                                                                                                                                 
riscv64-unknown-elf-objcopy -O ihex testblink.elf testblink.hex                                                                                                                                    
gcc -o busbflash busbflash.c -ludev -I../../ch32v003fun/minichlink                                                                                                                                 
make -C ../../ch32v003fun/minichlink all                                                         
make[1]: Entering directory '/home/username/ch32v003/rv003usb/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.                                                           
make[1]: Leaving directory '/home/username/ch32v003/rv003usb/ch32v003fun/minichlink'
./busbflash                                     
HD: (nil)                                       
make: *** [../../ch32v003fun/ch32fun/ch32fun.mk:292: cv_flash] Error 252        
```                

I noticed that the folder structure is different to the video. 
Also the Makefile seems broken, i.e. it doesn't compile out of the box after git clone. I had to change paths and add the `TARGET_MCU?=CH32V003`. The Makefile had the same folder structure as in the video but it is different in the git repo.

Also my compilation parameters seem to differ a bit to those from the video. I don't know why and if that's problematic. I noticed on my system also calls `cc` and not just `riscv64-unknown-elf-gcc`.

Any ideas?

------------
My system:
`Linux mycomputer 6.1.131-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu Mar 13 15:02:58 UTC 2025 x86_64 GNU/Linux`

wohltat, 2025-04-19:
After writing the message i realized that the usb device was not found for my user but with `sudo`. Guess it's a missing udev rule or something.

now i get:
``` bash
> sudo ./busbflash testblink.bin
HD: 0x558b54cfa5e0
Read 40 bytes
Send feature report: 128
Read: 129
```

wohltat, 2025-04-19:
Awesome, it blinks now!

The testblink example seems like an older in-process-version, or am i missing something?
I had to add a loop to make it blink like the name suggests.

```
#include "ch32fun.h"

//XXX TODO Make this so it can "run from main"
void Scratchpad( uint8_t * buffer, uint32_t * keeprunning )
{
    GPIOC->CFGLR = (GPIO_Speed_50MHz | GPIO_CNF_OUT_PP)<<(4*0);
    
    while (1)
    {
        if( SysTick->CNT & (1UL << 24) )
        {
            GPIOC->BSHR = 1<<0;
        }
        else
        {
            GPIOC->BSHR = 1<<16;
        }
    }
}

```

There are still some open questions for me.
- is the scratchpad the user code area (non bootloader)?
- Why is its default size 128? Why not something else like the 16KB of the flash? I guess its for speed reasons and it doesn't yet handle bin file sizes dynamically?

cnlohr, 2025-04-19:
1. Scratchpad is just where code is written by the host that will be executed.
2. It needs to be something "bite-sized" because natively you can write up to 64-bytes at a time to each sector, you can write the code needed to flash a sector, and the 64-bytes for the sector into that space.
3. You do not want it to be 16kB, because (1) there is not enough RAM.  And (2) you would need to make ALL of your USB transfers that big because HID transfers are of fixed sized (only enforced on Windows).

cnlohr, 2025-04-19:
If this answers all your questions please close.

wohltat, 2025-04-20:
My initial problem is solved so i closed the issue. But some other questions came up later:

1) As i understand the boot mode can only be triggered by a power cycle, correct? Is there a possibility for a software reset to boot mode like for arduino or esp32? That would make it super convenient.
2) The reset button is not enough to restart to the bootloader on my board. Is that normal?
3) As i understand you, Scratchpad is just a place for transfered chucks of code. That means the complete code can fill up more. The full 16KB or MEMORY (default=512)?
I'm a bit confused because because i tried to put some code in the `Scratchpad` function and got the error:

```
ld: testblink.elf section `.init' will not fit in region `SCRATCHPAD'
...
Memory region         Used Size  Region Size  %age Used
      SCRATCHPAD:         136 B        128 B    106.25%
          MEMORY:          0 GB        512 B      0.00%
```
4) How to use functions like `Delay_Ms( 500 )`? When i use them i get Linker error
```
undefined reference to `DelaySysTick'
```
In the "normal"(ch32fun) blink example i don't get a linker error.

cnlohr, 2025-04-21:
1. I believe @monte-monte has a feature to allow booting-into-bootloader.
2. Depending on fuse settings, the RESET pin is typically just mapped to GPIO.  But, it doesn't have to be.
3. You can write into more of the space, but the scratchpad is specifically 128 bytes because that was the trade off we made with regard to HID requests.  Because in Windows, HID requests must be of fixed sizes.
4. Only ultra specific functionality should be used with the scratchpad.  You should be flashing the part with whatever programs you want to run using ch32fun.

monte-monte, 2025-04-21:
@wohltat in the latest update to ch32fun there is now functionality in minichlink to reboot from rv003usb to bootloader automagically. You would need the relatively fresh copy of rv003usb too.
Reset button works weird, I prefer to disable it.

monte-monte, 2025-04-22:
@wohltat I never tested testblink. Could you try with ``demo_hidapi`` or ``demo_terminal``.
Are you on linux or windows?

wohltat, 2025-04-22:
> Are you on linux or windows?

> My system:
> `Linux mycomputer 6.1.131-1-MANJARO #1 SMP PREEMPT_DYNAMIC Thu Mar 13 15:02:58 UTC 2025 x86_64 GNU/Linux`

-----------------------

Yeahhh i seem to work now. So cool!!!

First `demo_hidapi` had a similar problem. Flashes once an then gets kinda stuck.
``` bash
> ../ch32v003fun/minichlink/minichlink -w demo_hidapi.bin flash -b
VID:0x1209, PID:0xb003
Found B003Fun Bootloader
Halting Boot Countdown
Interface Setup
Writing image
..........................................................................................................................................................................................................................................................................
Image written.
Booting
> ../ch32v003fun/minichlink/minichlink -w demo_hidapi.bin flash -b
VID:0x1209, PID:0xb003
Error: Could not initialize any supported programmers
```

Then i noticed that its not the same vid/pid:
```
[313762.446494] usb 1-1.2: USB disconnect, device number 35
[313762.657363] usb 1-1.2: new low-speed USB device number 36 using ehci-pci
[313762.759497] usb 1-1.2: New USB device found, idVendor=1209, idProduct=b003, bcdDevice= 0.02
[313762.759517] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[313762.759524] usb 1-1.2: Product: rv003usb
[313762.759529] usb 1-1.2: Manufacturer: cnlohr
[313762.759533] usb 1-1.2: SerialNumber: NBTT
[313762.763120] hid-generic 0003:1209:B003.0171: hiddev97,hidraw4: USB HID v1.10 Device [cnlohr rv003usb] on usb-0000:00:1a.0-1.2/input0
[313765.774506] usb 1-1.2: USB disconnect, device number 36
[313765.987214] usb 1-1.2: new low-speed USB device number 37 using ehci-pci
[313766.090422] usb 1-1.2: New USB device found, idVendor=1209, idProduct=d003, bcdDevice= 0.02
[313766.090431] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[313766.090434] usb 1-1.2: Product: RV003 Custom Device
[313766.090436] usb 1-1.2: Manufacturer: CNLohr
[313766.090438] usb 1-1.2: SerialNumber: CUSTOMDEVICE000
```
Since i don't have that in my udev rules yet it didn't run without sudo. With sudo it did:
``` bash
> sudo ../ch32v003fun/minichlink/minichlink -w demo_hidapi.bin flash -b
WARNING: You are not in the plugdev/dialout group, the canned udev rules will not work on your system.
VID:0x1209, PID:0xb003
Trying to reboot into bootloader
Sent magic packet
Found B003Fun Bootloader
Halting Boot Countdown
Interface Setup
Writing image
..........................................................................................................................................................................................................................................................................
Image written.
Booting
```

-------------
`testblink` example still has another problem. `sudo` didn't fix it. The USB-Connection is lost there.

---------------

`demo_terminal` didn't compile. I think i better open a new issue for that.

wohltat, 2025-04-22:
Is it necessary to have a different `pid` for it to work?

> [313762.759497] usb 1-1.2: New USB device found, idVendor=1209, idProduct=b003, bcdDevice= 0.02

> [313766.090422] usb 1-1.2: New USB device found, idVendor=1209, idProduct=d003, bcdDevice= 0.02

monte-monte, 2025-04-23:
Yes, we want to distinguish bootloader from normal rv003usb or anything else.

arduino12, 2026-03-14:
> I don't know if i'm doing something wrong and how it is intended to be used.
> 
> It's working partially. Both rv003usb and ch32fun are the latest version.
> 
> * After connection it's in the bootloader and i can flash with both minichlink and busbflash at least once.
> * if i flash with `minichlink -w blink.bin flash -b` to flash and reset, it does that, but ends up in a state where i can't flash again ("USB disconnect")
> * if i flash the testblink example (`make clean all flash`, i.e. with `busbflash`) it runs and stays in a responsive state. But i cannot flash run it again. Maybe it's flashed but i don't see an effect.
> * if i flash the testblink example (`busbflash`) and then use `minichlink -b` i get  back to the bootloader. Now i can flash and run it again. This can be repeated multiple times.
> 
> It feels to me i'm still missing something.

I have a simular problem - the bootloader only works once - and not always..
I use Win11 + PlatformIO to compile and upload the bootloader and later the demo_hid code.
I use the latest version of this repo, also updated the `%USERPROFILE%\.platformio\packages\tool-minichlink\minichlink.exe`.
I got same results while using both Link-E and ESP32-S2 programmers to upload the bootloader.
The demo_hid works when I upload it via my programmers - so its not a USB hardware problem.

I run `pio run -t upload -v` and see `minichlink -w .pio\build\demo_exti\firmware.bin 0x08000000 -b`-
so the bootloader is being flashed to 0x08000000 (instead of 0x1FFFF000 ?) when using the PlatformIO.
I run: `minichlink.exe -w rv003usb\.pio\build\bootloader\firmware.bin 0x1FFFF000`-
after that - I can upload the demo_hid multiple times without the programmers - via the USB bootloader!
USB upload doesn't work every time - so I just unplug and replug and try again until I see `demo_composite_hid  SUCCESS`.
Maybe need to change the `platformio.ini [env:bootloader]` to upload to 0x1FFFF000?