cnlohr/rv003usb #98: Why doesn't 5V power work for USB?

egormanga, 2025-05-25:
> … are used to run the CH32V003 at 3.3V. For USB it should not be run beyond 3.6V.

What causes the requirement of going not higher than 3.6V? Is it in some spec, or does the V003 run too fast/unstable on it?

BogdanTheGeek, 2025-05-27:
@egormanga USB data lines are 3.3v. You can run it off 5V VDD if you add clamping diodes to data lines.

egormanga, 2025-05-27:
Thanks for the hint! But aren't USB data lines a differential pair? Why would the voltage matter then?

jjflash65, 2025-05-30:
Perhaps you want to look here: https://www.jjflash.de/ch32v003/v003diy_programmer.jpg

jjflash65, 2025-05-30:
Forgoten to write: Use Z-Diodes with (very) small P_tot, lower-equal than 0,25W, because the characteristic curve in reverse direction isn't good with such low-voltage Z-Diodes. With more powerfull Z-Diodes it is possible to have larger voltages than 3.3V

cnlohr, 2025-06-02:
You can certainly do that.  It all boils down to USB requires the data lines tolerate 5V on them, but they are not guaranteed to work outside of the 0..3.6v range.

closed 2025-06-02 by cnlohr