cnlohr/rv003usb #102: USB-MIDI Library (Based on the Mixtela Awesome project "Smallest USB-C MIDI Synth")

NoNamedCat, 2025-07-01:
[demo_midi.zip](https://github.com/user-attachments/files/20997273/demo_midi.zip)

This MIDI library is a derivative work based on the brilliant SMSC project by Mitxela. It adapts Mitxela's original concepts to enable MIDI functionality on the CH32V003 microcontroller using the RV003USB SDK. All core ideas and foundational code are credited entirely to Mitxela

For more details on the original project, visit: https://mitxela.com/projects/smsc

cnlohr, 2025-11-18:
Do you have contact with mitxela that you could see if they're interested in upstreaming?

NoNamedCat, 2025-11-19:
I can try opening an issue on https://mitxela.com/projects/sms and ask for permission and suggestions on what might be the best option that he would be happy with.

I’m going to reference this conversation in that issue so he can see this as well.

Is this okay for you?

NoNamedCat, 2025-11-22:
Sorry, I’ve been disconnected because of work (business trips). Mixtela already replied to the issue I posted. Please check that issue in the GitHub repository of this project.

https://github.com/mitxela/smsc/issues/1

cnlohr, 2025-11-22:
I think if we were doing that, let's turn this into a proper PR.

NoNamedCat, 2025-11-22:
It was never meant to be a proper PR; it was made in a hurry for a friend that was thinkering with MIDI devices. Need to change a lot