cnlohr/rv003usb #106: CH32V003F4U6 flashes but USB Device not Recognized

LimesKey, 2025-08-11:
```zsh
/Dow/h/rv003usb-master/bootloader ❯ sudo make                                                
make -C ../ch32v003fun/ch32fun/../minichlink all
make[1]: Entering directory '/home/ryan/Downloads/hackyholiday/rv003usb-master/ch32v003fun/minichlink'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ryan/Downloads/hackyholiday/rv003usb-master/ch32v003fun/minichlink'
../ch32v003fun/ch32fun/../minichlink/minichlink -a -w bootloader.bin bootloader -B
Found WCH Link
WCH Programmer is LinkE version 2.10
Already Connected
Full Chip Type Reply: [4] 81-55-01-01-00-00-00-00-00
Unknown chip fallback
No detected chip.
Setup success
Interface Setup
Writing image
Chip Type: 000
Unknown chip type.  Report as bug with picture of chip.
Vendored: 00000000
marchid : 00000000
HARTINFO: 00000000

Image written.
```
I flashed it using Linux with a USB-C WCH Programmer, and I'm getting the USB error in Windows. Is there a way to verify the firmware was flashed correctly? And is this `Unknown chip type.  Report as bug with picture of chip.` concerning?

LimesKey, 2025-08-11:
I updated the WCH Link E firmware from `2.10` to `2.18` and now I get this,
```zsh
/Dow/h/r/ch32v003fun/minichlink ❯ sudo ./minichlink -i                                           03:06:49 AM
Found WCH Link
WCH Programmer is LinkE version 2.18
Already Connected
Full Chip Type Reply: [4] 81-55-01-01-11-00-00-00-01
Unknown chip fallback
No detected chip.
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on op (DMABSTRACTS = ffffffff) (-1) (Other Error) DMSTATUS: ffffffff
Retrying
Fault on setup -9
Could not setup interface.
```

cnlohr, 2025-08-12:
Ah, sorry, I meant `make unbrick` not unlock. 

That makes more sense.  That it's a real issue.

Considering the errors in the second set, I think that it's a real issue. 

1. Which processor are you connecting? There is no SWC on the ch32v003.  SWDIO should be connected to PD1
2. What voltage are you giving the processor?  Can you confirm it's 3.3V?
3. Generally the crystal should not be an issue.  All our examples by default use the internal RC oscillator.

If you're interested in joining the Discord to get immediate help with this, feel free to DM me at `cnlohr` on Discord.

closed 2025-08-26 by LimesKey